hawku/TabletDriver #127: CTH 480 cursor Jumping while hovering or lifting up and down purely on dx or dy cursor position but resets.

Tioaoa, 2018-04-23:
youtube video https://youtu.be/MvzciDAf8Q0

## Hardware/software information
Windows Version: windows 10 64

Tablet Driver version: 0.1.5

Tablet model: CTH 480

Tioaoa, 2018-04-23:
yes, pixel  jump distance from original position is i would say around 150px or 30px on either the literal dx or dy axis

it seems to do this if i lift up or down at a certain height.. say .5 to .6cm

Devocub, 2018-04-23:
If it happens only in osu! and okay in Windows is means it's osu! issue but not tablet driver issue.

Tioaoa, 2018-04-23:
I said yes to your reply.. which means it does happen outside of osu

Devocub, 2018-04-23:
Oh, sorry

Tioaoa, 2018-04-23:
and it is only with Hawk's tablet driver

Devocub, 2018-04-23:
I think hawku wall ask you for debug log. Check this #83

Tioaoa, 2018-04-23:
ok i have found the issue
it is pretty weird, but I think it maAaAaAY BE an easy fix.
when holding down one of the pen's built in key's it will cause that issue, however without the constant keypress it works just fine.

yes I have tried disabling key input through Tabletdriver.exe but this does not resolve it... it MIGHT be more of a hardware thing but pffff idk.. the only reason i found out what by taking of my rubber band from it.

imNP, 2018-04-24:
@Tioaoa Hey, I'm the person who opened up the issue referenced above and thought I'd give you some insight. I saw the video and when I was having the issue my cursor wasn't jumping nearly as bad as that it would either jump to the edge of the screen or make several small jumps here or there. I found that when I tilted my pen slightly is what caused it to trigger it but only at certain angles (at least on my device). I know this probably isn't what you want to here but getting a new tablet is probably going to be the only way you are going to be able to fix this issue. When I got my CTL-472 and tried the driver out again the issue was completely gone. If you want to stick Wacom Tablets so you can keep your play area and don't want to pay a whole lot of money for a 480 as they are becoming harder to come by (especially new or in good used condition) just get the 472 since it's pretty much identical in every way to the 480 except that the pen is a little lighter.

hawku, 2018-04-24:
@Tioaoa You could try add `DetectMask 0x20` to the "Commands" tab's bottom field.

That will the change how the position packets will be validated.
Hover height will be lower, but there might be less invalid cursor positions.

I have noticed that Wacom tablets will sometimes send invalid position packets when the barrel button and the pen tip is pressed down.
It seems that the `DetectMask 0x20` setting might also be the best choice for other Wacom tablets.
I will test that out on my tablets and will change the default values.

Tioaoa, 2018-04-24:
yeh that resolved about 95% of it but I dont know what a barrel button is >> 

the issue shows up at a certain range while holding down one of the side buttons(i assume that is the barrel button) but I should note that this doesnot happen with concurrent wacom drivers... maybe because there is a mask already idfk. i do not press down on the pen tip: I am a hover player.

Tioaoa, 2018-04-24:
I mean.. what you could do it set a distRange() through a while loop under a ranged instance of time(0.001s), to where it checks if the cursor moves waaay to quickly (inhuman reaction) but I know this is an expense on latency..

AuxiliusM, 2018-04-25:
i had this exact same problem and can confirm that DetectMask 0x20 helps with about 95% of it

Tioaoa, 2018-04-28:
but the thing is I cant use detectmask becuase my muscle memory is all about breaking that detection point, so when i play it just *lags* becuase i go beyond the detection mask >>

AuxiliusM, 2018-05-30:
had to realise that 2 range is just 2 short

closed 2018-12-12 by hawku

Alejandroski33, 2019-10-01:
ITS THE CABLE!!, i justi changed my cable and now it doesn't have that problem