hawku/TabletDriver #128: Multiple tablet settings

Dantoxism, 2018-04-25:
It's not as much of an issue as it is an inconvenience, however it would be nice if you could add something like there is in the original tablet drivers where you have a different preset for different programmes, would be very helpful to play osu! and still be able to draw with the same driver :)

SuperTilted, 2018-04-27:
This is similar to #107, which Hawku has addressed:

> I'm planning to add profiles and it will need large changes to the GUI code, so I might have to rewrite most of it.

For the time being, the best thing you can do is make use of the driver's import/export function. Just export the settings you want for each program and then just import whichever one you need to use. You'll need to have v0.1.1 or later to be able to do this, though.

hawku, 2018-05-02:
Duplicate of #50

closed 2018-05-02 by hawku