hawku/TabletDriver #136: Driver uses both of my monitors

AldumSK, 2018-05-05:
It worked all fine until one day, all of the sudden the driver used both of my monitors for my tablet area,I´ve set it as my main but that didn´t help . Only unplugging my second monitor helped. I´ve tried uninstalling and reinstalling it but it didn´t work, and I tried the older versions with no effect . pls help ;-;
(I also tried setting up the desktop size but it didnt change anything)

screenshoots: http://prntscr.com/je72nh
                      
## Hardware/software information
Windows Version: Win10 Pro

Tablet Driver version: I tried from ver0.1.2-0.1.5

Tablet model: CTL-471

AldumSK, 2018-05-06:
Ok now anther problem just occurred today, now it doesn´t use my tablet area at all it uses all of it...

AldumSK, 2018-05-06:
Another update...
everthing works now after I just installed my original wacom driver and just closed the application, and tzhen running the GUI again. 
[image]

hawku, 2018-12-12:
@AldumSK Maybe Windows 10 automatically installed some Wacom drivers and installing the correct Wacom drivers fixed the problem.

closed 2018-12-12 by hawku