hawku/TabletDriver #138: Star g430s only quater of screem

Felkiiee, 2018-05-07:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.5

Tablet model: Star g430s

The only portion where i can move my cursor with my tablet is the top left quarter of my screen. Why is this?

Devocub, 2018-05-07:
Check #129

sssata, 2018-05-24:
I found a workaround by editing the tablet.cfg file in the config folder.

Comment out the existing settings for the G430 and add this:

`#
# XP-Pen G430s
#
Tablet 0x28BD 0x0075 0xFF0A 0x0001
Name "XP-Pen G430s"
ReportId 0x02
ReportLength 8
DetectMask 0x80
MaxX 20480
MaxY 15240
MaxPressure 8191
Width 102.4
Height 76.2
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00`

Note that you'll have to revert this if you want to use the G430.

frodriguez96, 2018-07-04:
Here are some better config settings using a 1:200 ratio between width -> max x and height -> max y, just like the G640 config. 
Also the tablet width is actually 101.6mm, this should properly cover the whole tablet area and display.

Just like above, comment out the G430 lines or replace them completely with the G430s since they both are recognised as the same tablet.

```
#
# XP-Pen G430s
#
Tablet 0x28BD 0x0075 0xFF0A 0x0001
Name "XP-Pen G430s"
ReportId 0x02
ReportLength 8
DetectMask 0x80
MaxX 20320
MaxY 15240
MaxPressure 8191
Width 101.6
Height 76.2
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00
```

gonzup, 2018-11-11:
Hello, i have the same problem like the post owner, and i edited that thing you mencioned but i have no response, was the same, bc keep everything the same..

hawku, 2018-12-10:
@Felkiiee and @sssata Please, check the issue #301

closed 2018-12-12 by hawku