hawku/TabletDriver #141: add support for wireless accessory kit?

xyning, 2018-05-11:
this driver works well with my CTH-680 when using usb cable  
but it doesnt seem to work on wireless mode  
is it possible to support it?

## Hardware/software information
Windows Version: Windows 10 1709

Tablet Driver version: 0.1.5

Tablet model: CTH-680

Kashouryo, 2019-04-08:
Same issue here. CTH-690 with wireless kit won't detect anything. Wired connection works fine.

xyning, 2019-04-08:
```  
# Wacom Wireless Accessory Kit CTH-680 (Wacom drivers installed)
HIDTablet 0x056A 0x0084 0xFF00 0x000A
Name "Wacom WAK CTH-680 (Wacom drivers installed)"
ReportId 0x02
ReportLength 11
DetectMask 0x40
MaxX 21600
MaxY 13500
MaxPressure 1023
Width 216.0
Height 135.0
TabletFormat WacomDrivers  
```
this works with my wireless CTH-680 with wacom driver installed...  
seems 690 uses the same wireless accessory kit as 680...so maybe it ll also work..?
btw u may need to change x/y/w/h/pressure to CTH-690's actual values