hawku/TabletDriver #142: Possibility to fix input delay while hovering with CTL-490?

pasi123567, 2018-05-12:
Since launch of the CTL-490 everyone knows it has input delay if you hover with it. This is still the case with this driver but it definitely decreased in amount. But as a hover player this bothers me quiet a bit. Thanks to this driver this is now the tablet with the least amount of input delay that I own. My G430 now feels like the CTL-490 while hovering, so I was suprised when I tried dragging with this one because it seemed like I move faster than time. 
Is there any way to fix this issue? I contacted Wacom multiple times but all they say is "They are working on it" <---- That was a year ago lol
Maybe you know a way? They said to me its software related.

## Hardware/software information
**Windows Version:** 
Windows 10 Home
**Tablet Driver version:** 
v0.1.5

CSharpProgramming, 2018-05-13:
It's not software related its hardware.. even if you extracted 100% raw inputs from the tablet it would be just as smoothed

hawku, 2018-05-21:
@pasi123567 It's a hardware/firmware issue. There isn't a way to disable the hover smoothing/delay.

closed 2018-05-21 by hawku