hawku/TabletDriver #146: System Error

davidew, 2018-05-18:
Windows 10 Home (1803)

v0.1.5

Wacom Intuos CTL 480

When starting TabletDriverService.exe I get a pop up sayin "System Error" followed by "The execution of the code could not continue because MSVCP140.dll. This problem can be solved reinstalling the program." And one that says the same except instead of MSVCP140.dll it's VCRUNTIME140.dll .

I've tried reinstalling and redownloading several times, but no changes.
I uninstalled my Wacom drivers.

davidew, 2018-05-18:
Edit: I fixed it by installing vc_redist. Seems that was the solution.

closed 2018-05-21 by hawku

hhhhose, 2018-07-12:
what is vc_redist? im having the same issue but have no clue what the solution you listed is

Banbeucmas, 2018-07-13:
@hhhhose
It is lised in the Readme.md of the software.
Just search it up. It is a kind of a library for the software to work.