hawku/TabletDriver #154: Huion 1060Pro+ incorrect default area

AdventGit, 2018-05-24:
I've manually added my Huion 1060Pro+ but I cannot figure out how to properly adjust the default area beyond what I currently have:

`Tablet 0x256C 0x006E 0x000D 0x0002`
`Name "Huion 1060PRO+"`
`ReportLength 8`
`MaxX 25400`
`MaxY 15875`
`MaxPressure 8191`
`Width 254.00`
`Height 158.75`

It uses only half of my height and 3/4 of my width as well as being slightly to far to the left. Maybe I don't understand what the difference is between MaxX/Y vs. Width/Height?

Some more information from console:

`>> CheckTablet`
`>> DesktopSize 1920 1080`
`Desktop size = (1920.00 px x 1080.00 px)`
`>> ScreenArea 1920 1080 0 0`
`Screen area = (w=1920.00, h=1080.00, x=0.00, y=0.00)`
`>> TabletArea 254 158.75 127 79.38`
`Tablet area: (254.00 mm x 158.75 mm X+127.00 mm Y+79.38 mm)`
`>> Rotate 0`
`Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]`
`>> Mode Absolute`
`Output Mode = Absolute`
`>> ButtonMap 0 0 0`
`Button Map = 0 0 0 0 0 0 0 0 `
`>> Smoothing 0`
`Smoothing = off`
`>> SmoothingInterval 2`
`Smoothing Interval = 2 (500.00 Hz, 0.00 ms, 1.000000)`
`>> Info`

`Tablet: Huion 1060PRO+`
`  Width = 254.00 mm`
`  Height = 158.75 mm`
`  Max X = 25400`
`  Max Y = 15875`
`  Max Pressure = 8191`
`  Click Pressure = 0`
`  Keep Tip Down = 0 packets`
`  Report Id = 00`
`  Report Length = 8 bytes`
`  Detect Mask = 0x00`
`  Ignore Mask = 0x00`
`  Button Map = 0 0 0 0 0 0 0 0 `

`Area:`
`  Desktop = 1920px x 1080px`
`  Screen Map = 1920px x 1080px @ X+0px, Y+0px`
`  Tablet =  254.00mm x 158.75mm @ Min(127.00mm, 79.38mm) Max(381.00mm, 238.13mm)`
`  Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]`

`HID Device: Vendor: '' Product: '1060PRO'`
`  Vendor Id: 0x256C, Product Id: 0x006E`
`  Usage Page: 0x000D, Usage: 0x0002`
`  FeatureLen: 0, InputLen: 8, OutputLen: 0`

## Hardware/software information
Driver version: 0.1.5
Windows version: Microsoft Windows NT 6.1.7601 Service Pack 1
Windows product: Windows 7 Ultimate
Tablet model: Huion 1060Pro+

Ikikaera, 2018-06-26:
Hey, I use an H610Pro and had the same issue. 
I copied what you had and it almost worked for me except that it couldn't reach the edge of my screen by quite a bit, both horizontally and vertically.

I had a friend help me fix it, it took quite a bit of testing around but in the end what needed to be changed was MaxX and MaxY.
I had to change it to:
MaxX 40000
MaxY 25000

Perhaps those could work for you too as our tablets seem to be quite similar in a lot of aspects. 
Also make sure to click on "Restart Driver" every time you make changes, or else it won't apply.

riley-badour, 2019-01-18:
i thinks i just called 1060 pro but at #90 we have  the proper code that works with the vmuti drivers
feel free to test it out