hawku/TabletDriver #162: Huion/INSPIROY Q11K Support?

octoberU, 2018-06-02:
Hi, is there anyway I could request support for my tablet? I have been using the huion drivers which are the same as the H420 but I couldn't get TabletDriver0.1.5 to recognize my tablet. Is there anyway I can provide info to get my tablet running on this driver?

## Hardware/software information
Windows Version:  Windows 10 Pro

Tablet Driver version: 
Huion v13.15.35
TabletDriver 0.1.5

Tablet model:  Huion/Inspiroy Q11K
https://www.huiontablet.com/inspiroy-q11k.html

hawku, 2018-12-12:
@octoberU Try the latest version. It should detect the tablet as an unknown Huion tablet and will provides more information to the `startuplog.txt`

octoberU, 2018-12-14:
This it what I found in the startup log:
[2018-12-14 17:00:53.270] >> Start
[2018-12-14 17:00:53.277] TabletDriver started!
[2018-12-14 17:00:53.277] [STATUS] TABLET Unknown Huion Tablet! Check startuplog.txt!
[2018-12-14 17:00:53.329] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2018-12-14 17:00:53.329] [STATUS] WIDTH 100.00000
[2018-12-14 17:00:53.329] [STATUS] HEIGHT 100.00000
[2018-12-14 17:00:53.329] [STATUS] MAX_X 10000
[2018-12-14 17:00:53.329] [STATUS] MAX_Y 10000
[2018-12-14 17:00:53.329] [STATUS] MAX_PRESSURE 2047
[2018-12-14 17:00:53.329] >> Log Off

Here's what it added to the cfg: 
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
Name "Unknown Huion Tablet! Check startuplog.txt!"
#CheckString 1 "???"
#CheckString 2 "???"
#CheckString 6 "???"
#CheckString 200 "???"
#CheckString 201 "???"
ReportId 0x07
ReportLength 8
DetectMask 0x80
MaxX 10000
MaxY 10000
MaxPressure 2047
Width 100
Height 100
InitStrings 100 200

https://www.huiontablet.com/inspiroy-q11k.html

How would I go about configuring my tablet to work with the driver? I tried mimicking some of the other tablets with their configs but ended up going nowhere. Any help is appreciated.

octoberU, 2018-12-14:
I managed to get it working. Here is my final config if you want to add it to the next release for others to use. Still trying to figure out what filters work best. Thank you for helping out and updating the driver.
```
#
# Huion Inspiroy Q11K
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
Name "Inspiroy Q11K"
CheckString 201 "HUION_T164_170321"
ReportId 0x08
ReportLength 12
DetectMask 0x80
IgnoreMask 0x60
MaxX 55880
MaxY 34920
MaxPressure 8191
Width 279.4
Height 174.6
InitStrings 100 200
```

closed 2018-12-14 by octoberU

hawku, 2018-12-15:
@octoberU Thanks for the tablet configuration.
That will be added to the driver in the next release.