hawku/TabletDriver #167: issue w/ installing vmulti driver

yeezyseasom21, 2018-06-10:
I have an issue with installing the vmulti driver 
[image]
i tried to unzip the driver to a shorter path but it didn't work. 
[image]
this is what happens when i try to run tabletdrivergui.exe
I have installed the drivers following the steps, and also tried to re download the libraries provided

## Hardware/software information
Windows Version: 7 ultimate 64 bit

Tablet Driver version: 0.1.5

Tablet model: ctl 480

YChvanov, 2018-06-10:
You are not to copy-paste "install_vmulti_64.bat" script in cmd, you have to rmb it in explorer and select 'Run as administrator'

yeezyseasom21, 2018-06-10:
I didn't copy paste the script in cmd, I just ran "install_vmulti_driver.bat" as administrator in the TabletDriverV0.0.15 folder and this is what I got: 
[image]

closed 2018-12-12 by hawku