hawku/TabletDriver #169: problem on dual screen

andyrkn, 2018-06-10:
## Hardware/software information
Windows Version:  win10

Tablet Driver version: latest

Tablet model: ctl 480 wacom

Laptop(gtx 950m at 768p) , got a 1080p plugged into the laptop on which i play, when trying to set the area on the second screen , it only registers half of it regardless of what i set. can only fix this by disabling the laptop screen trough windows, restarting the driver connecting the tablet and then turning on the screen again

simo3003, 2018-06-13:
Please post a screenshot of your settings.

andyrkn, 2018-06-13:
https://puu.sh/AECKR/a3860438d6.png

Devocub, 2018-06-13:
You said you use latest version of driver but you doesn't

andyrkn, 2018-06-13:
https://puu.sh/AED2I/0e9983c6c2.png

problem still persists

hawku, 2018-08-12:
You could try to disable the automatic desktop size and set the desktop width and height to match the external monitor.

andyrkn, 2018-08-20:
it doesnt work,

hawku, 2018-12-12:
@andyrkn Problem might be with how Windows handles absolute mouse positioning. Some displays just can't be used with absolute mouse.
I can't replicate that problem, so there is no solution for that yet.

closed 2018-12-12 by hawku

andyrkn, 2018-12-28:
@hawku have you tested with two monitors off different resolution?