hawku/TabletDriver #171: Alot of input lag while hovering with pen

Reflexayy, 2018-06-10:
There is alot of input lag when i hold the pen above my tablet, the input lag goes away when I either put my pen down on the tablet or move the pen off the tablet entirely. I recently updated windows and this started to happen, yet it does not happen on the company's tablet drivers.

## Hardware/software information
Windows Version: 
Microsoft Windows 10 Version 1803 (OS Build 17134.1)
Tablet Driver version: 
0.1.5
Tablet model: 
XP-PEN G430S

halbGefressen, 2018-06-21:
G430S is not officially supported yet - I hope support will be added ASAP (because I also own the G430S)

closed 2018-08-12 by hawku

DizzasTeR, 2019-11-21:
Having this exact issue on G430s while it is supported now, had to apply the fix from https://github.com/hawku/TabletDriver/issues/138 to make it detect but now getting huge fps drops and latency when hovering