hawku/TabletDriver #174: Vikoo hk708s (Different brand) Support

s0m3guy2004, 2018-06-15:
Can you please add support for a different Tablet brands?

##Logs:

2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0x0000, Usage: 0x000E
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 3, InputLen: 0, OutputLen: 0
2018-06-15 23:31:44 [HIDDevice] [DEBUG] 
2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-06-15 23:31:44 [HIDDevice] [DEBUG] 
2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0001
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 65, OutputLen: 65
2018-06-15 23:31:44 [HIDDevice] [DEBUG] 
2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0002
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 33, OutputLen: 65
2018-06-15 23:31:44 [HIDDevice] [DEBUG] 
2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'PenTablet' Product: 'Simulated HID MiniDriver'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x5543, Product Id: 0xFFEF
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-06-15 23:31:44 [HIDDevice] [DEBUG] 
2018-06-15 23:31:44 [HIDDevice] [DEBUG] HID Device: Vendor: 'PenTablet' Product: 'Simulated HID MiniDriver'
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Vendor Id: 0x5543, Product Id: 0xFFEF
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0002
2018-06-15 23:31:44 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0

## Hardware/software information
Windows Version: Windows 7 64-bit

Tablet Driver version: 8.1.2016.1002

Tablet model: Vikoo hk708s

##Stock driver
It could be useful.
https://drive.google.com/open?id=1OttG965Qj8C5lGJL2S8FJB8DMjqcIORe

nijunnijun, 2020-07-02:
Looking forward to have Vikoo hk708 supported too. Thanks!!!

EduChappie, 2024-04-25:
I think i was the only person in the world who had a problem to find the driver of this table (Vikoo HK708s)
but anyway, I want to help me in the future or someone else who has had problem to find this driver, so
there is the link: [VikooPen 708s](https://drive.google.com/file/d/1XrmsT2szre-lCXNKIFj_8XjDOe5eYZZI/view?usp=sharing)

I know there are some videos in Chinese that speak about this table, but I don't speak Chinese
I hope help someone

s0m3sushi, 2024-04-25:
It's been like 6 years lmaooo

ololichopopo, 2025-07-20:
i have no idea how to reply to educhappi i just had to make an account to thank you lol