hawku/TabletDriver #177: Very shaky

Rexzosu, 2018-06-20:
https://www.youtube.com/watch?v=Q44rPRGT10Y
^This is with hovering, stationary it still shakes but a little less.
I read that other people have a lot of shakyness, but I also see other streamers using custom driver and their cursor is as still as if they were using mouse. Is there any help/insight you can give?
Thanks for the amazing driver regardless <3 

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.5

Tablet model: CTH-480

Devocub, 2018-06-20:
https://github.com/hawku/TabletDriver/wiki/FAQ#q-why-is-the-cursor-shaking
https://github.com/Devocub/TabletDriver

Rexzosu, 2018-06-20:
Is that considered small amount?

Rexzosu, 2018-06-20:
Tried your driver and it fixed the shaking, thank you so much <3

closed 2018-06-20 by Rexzosu