hawku/TabletDriver #178: Does the CTH-690 work with the driver?

jophes123, 2018-06-21:
Is there a driver for the CTH-690 by any chance? I can't even install the vmulti driver, it says http://prntscr.com/jvos7i and im getting this in the console.. http://prntscr.com/jvvpov

## Hardware/software information
Windows Version: 10

Tablet Driver version: V0.1.5

Tablet model: CTH-690/K3-CX

hawku, 2018-08-12:
The driver should detect Wacom CTH-690.
Try to unzip the driver to different directory. For example: C:\Temp\TabletDriver

Miikk86, 2018-08-18:
I have the same tablet (CTH-690), using the same OS, and the same TD ver. and for me it works completely fine. Have you tried using this driver in other computer? It could be some machine issue...

closed 2018-12-12 by hawku