hawku/TabletDriver #182: Xp-pen G640

iberius, 2018-06-22:
Is there a way you could make the way you control the tablet area for the Xp-pen G640 the same as the software? cuz i cant fine a converter and i cant get it to be the same

## Hardware/software information
Windows Version: 

Tablet Driver version: 0.15

Tablet model: G640

okicarde, 2018-08-03:
Basically not. Anyway, why you want to do that ?

YChvanov, 2018-08-04:
Max area in PenTablet is 629x393 and in TabletDriver it's 160x100. Dividing first by second gives us that PenTablet area is bigger by 3.93 times. With this you can convert area size.
Example: input from PenTablet - w109 h70, output to TabletDriver - w27.735369 h17.8117048.
Probably position can be converted the same.
(yeah, bad English)