hawku/TabletDriver #185: "ERROR! Can't start the driver!", "Access is denied"

iOppy, 2018-06-28:
As the title states I my TabletDriver does not work at all and displays that error in the console ever since I tried updating it from the original version that I have had. I have tried everything, uninstalling, reinstalling, restarts, etc. I would appreciate some assistance so that I can get my drawing tablet back up and running again!

Thanks,

~Oppy

**## Hardware/software information**

**Windows Version:** Windows 10

**Tablet Driver version:** TabletDriverV0.1.5

**Tablet model:** Wacom Intuos Pro PTH-451

Banbeucmas, 2018-06-29:
May be open the TabletDriverGUI as Administrator works?

iOppy, 2018-06-30:
@Banbeucmas The application is already to "Run As Administrator" by default after I manually went into properties and set it up that way... Since I too thought that was the problem before I came here, but that was not the case and still received the same error after multiple attempts.

~Oppy

hawku, 2018-08-12:
@iOppy Where did you unzip the driver?
Try to unzip it to `C:\Temp\TabletDriver` directory

closed 2018-12-12 by hawku