hawku/TabletDriver #187: Shaky cursour even while dragging

Noxaniac, 2018-06-29:
So i just bought this tablet used been used 2-3 times before drawing not on osu
Downloaded the hawku drivers and has been a blast but since day one on the hawku drivers my cursour has always been shaky like even when im dragging its a little shake that doesnt go away and its horrible when i hover sometimes the cursour also jumps to the top or the bottom of the screen and teleports back to where i have the pen located on the tablet. I think i have the same issue as #83 
Here is a quick video https://youtu.be/1SDVB9x9K3c couldnt capture the random jump teleportation technique that my tablet does but anyways and here is the debug file [debug_2018-06-29_09_20_28.txt](https://github.com/hawku/TabletDriver/files/2150638/debug_2018-06-29_09_20_28.txt)

Sorry if this post is so messed up im not a native english speaker

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: Hawku 0.1.5

Tablet model: CTL-480

Devocub, 2018-06-29:
About shaking
#177 

About teleporting 
https://github.com/hawku/TabletDriver/issues/127#issuecomment-384027027

Noxaniac, 2018-07-02:
Alright i've fixed it, Solution for the shaking was to just download the Devocub Edition of the Driver and enable the smoothing filter and antichatter there

And the second fix was just to remove my huge ass 1cm width mouse pad. that i had over the tablet.. Huge thanks to Devocub!

closed 2018-08-12 by hawku