hawku/TabletDriver #188: Driver doesn't work newest windows version

33masterman33, 2018-06-30:
I have the newest version of Windows 10 installed and have everything the driver needs but neither this or the official one from XP-pen work this it shows the tablet driver is running and working but the settings do nothing. I try disabling a button then click apply and save but the button continues to work and its the same thing with all the other settings.

## Hardware/software information
Windows Version: Windows 10 Pro Ver 1803

Tablet Driver version: 0.1.5

Tablet model: XP-Pen G640

R-E-V, 2018-07-02:
Dude im having the exact same problem
im honestly about to just downgrade my windows

Devocub, 2018-07-02:
If official version doesn't work then it's not this driver problem at least

33masterman33, 2018-07-03:
@R-E-V I found a solution I downloaded an older driver and it works fine heres a link to the one I used https://www.dropbox.com/s/4er8rwibmdp5ihr/G430%20driver.zip?dl=0

33masterman33, 2018-07-03:
@Devocub I know but I was just noting it so hopefully they can fix it

R-E-V, 2018-07-03:
@33masterman33 is the older driver of this tablet UI or something else I'm at work and just saw this and I need to clean uninstall my Wacom drivers before install right ??

33masterman33, 2018-07-03:
@R-E-V oh well I driver I linked won’t work because I have a different brand but yeah what I would try doing is uninstalling the current driver then looking on wacoms site for an older version if you can’t find it there try a google search

R-E-V, 2018-07-03:
@33masterman33 do you have discord??

hawku, 2018-08-12:
@33masterman33 @R-E-V 
I just tested TabletDriver v0.1.5 with the XP-Pen G640 in Windows 10 1803 and everything seems to work without problems.
Have you tried the driver after installing the latest updates to Windows 10 1803?

closed 2018-11-27 by hawku