hawku/TabletDriver #189: Screen Mapping Issue (Dual Monitors)

PinkieBarto, 2018-06-30:
**I HAVE READ THE FAQ**

Okay so i'm trying to setup my tablet to only work on my main monitor in osu. I have set the screen map as well as desktop size to 1920x1080, however the cursor only reaches half of the screen vertically, and it doesn't reach the right edge fully horizontally. In osu I have disabled "map absolute raw input" but it didn't help. Any ideas what may be causing it? Is this a bug or have I made some stupid mistake somewhere?

I'll also attach a screenshot of the settings.

[image]

## Hardware/software information
Windows Version: 10 Education (10.0.17134 Build 17134)

Tablet Driver version: 0.1.5

Tablet model: XP-Pen G430s

Devocub, 2018-06-30:
Check #164

PinkieBarto, 2018-06-30:
Fixed. Thanks!

closed 2018-06-30 by PinkieBarto