hawku/TabletDriver #203: Error

Reddddddddd, 2018-07-11:
[image]

## Hardware/software information
Windows Version: 10

Tablet Driver version: 1.5

Tablet model: xp-pen g430

Banbeucmas, 2018-07-11:
Put the driver on another drive

Reddddddddd, 2018-07-11:
now im getting this
[image]

Banbeucmas, 2018-07-11:
Reinstall C++ 2015
As said on the main page.

> 
You might need to install these libraries, but usually these are already installed:

https://www.microsoft.com/net/download/dotnet-framework-runtime
https://aka.ms/vs/15/release/vc_redist.x86.exe

closed 2018-07-12 by Reddddddddd