hawku/TabletDriver #208: When I turn on windows ink, the pen pressure only works in the Windows Ink workspace and not in photoshop.

DomoJr, 2018-07-15:
This is really frustrating :(
I want to draw but i dont want shitty osu drivers either. Even if the windows ink mode is on, it doesnt do anything, just makes photoshop unusable. 

## Hardware/software information
Windows Version: 10

Tablet Driver version: 1.5

Tablet model: CTH-480

ok2094, 2018-07-26:
Not sure if this helps, but I use Photoshop without Windows Ink. These are my notes on how to make it work. 
https://www.evernote.com/shard/s549/nl/157201533/6763e11c-632b-48b5-9c2d-53ab905752eb?title=Photoshop%20Pressure%20Fix 

Had to make them since I reinstall my Windows every few years and forget how I got it to work lol 
I found this solution in one forum only.

ok2094, 2018-07-26:
Nevermind I just read that it doesn't have WinTab support, so my fix won't work. #84

hawku, 2018-12-11:
@ok2094 @DomoJr That Photoshop's "UseSystemStylus 1" setting should work.

closed 2018-12-11 by hawku