hawku/TabletDriver #209: Support for Linux?

Paussse, 2018-07-16:
Will that ever be a thing?

## Hardware/software information
Linux Kubuntu 

Tablet Driver version: 

Tablet model: XP-Pen G430

LonePerv, 2019-04-28:
I would like to have a support on Linux too, but since that much time passed since that question was asked, I assume, that it won't be possible ?

InfinityGhost, 2019-04-28:
It's possible, but it requires completely rewriting the GUI code because of its use of the .net framework for the GUI which is Windows only.

BrokenGale, 2019-08-19:
Is it possible to still get the driver to install and run, if I am willing to dive into the config files to change settings?

InfinityGhost, 2019-08-19:
> Is it possible to still get the driver to install and run, if I am willing to dive into the config files to change settings?

Your best chance is trying to run it in Wine. Modifying config files won't help it run on linux. It comes down to porting the TabletDriverService from Windows to Linux to get it to work.

Optimalgitit, 2020-03-18:
I'm pretty sure .NET is available for linux.

InfinityGhost, 2020-03-19:
> I'm pretty sure .NET is available for linux.

.NET Core, yes, but it doesn't support WPF on linux. Using something like mono maybe would work but its not a guarantee, plus the service would need to be written explicitly for linux anyway.

Ashe, 2020-04-13:
> > I'm pretty sure .NET is available for linux.
> 
> .NET Core, yes, but it doesn't support WPF on linux. Using something like mono maybe would work but its not a guarantee, plus the service would need to be written explicitly for linux anyway.

You don't necessarily need to build the GUI for linux - most linux users are used to CLI-based programs so I think you could get away with it. I don't know how your program runs but
1. If the drivers were able to run without a GUI based on a config file, people would have access to a nice low-latency driver on linux (never experienced anything other than the wacom ones so I could be using bad drivers I don't know)
2. If someone really did need the GUI, they could run the drivers in wine, generate a config file and then save it to be used every session. I feel like unless you're constantly changing settings, people will overlook the GUI thing after they've set things up! 

I don't really know much about drivers, but I could always give it my best and try and help bring this to linux after university is over for the semester. With osu!lazer coming soon(tm), cross platform drivers will be a huge bonus!

InfinityGhost, 2020-04-14:
> You don't necessarily need to build the GUI for linux - most linux users are used to CLI-based programs so I think you could get away with it. I don't know how your program runs but
>
>     1. If the [...]

Not to plug or anything but I'm currently working on a project like such all in .NET Core with Avalonia as the GUI framework (https://github.com/InfinityGhost/OpenTabletDriver)