hawku/TabletDriver #225: "devcon.exe failed" i get this issue, everytime i try to install my tablet drivers.

SgtCataclyst, 2018-08-07:
## Hardware/software information
Windows Version: 10

Tablet Driver version: V0.1.5

Tablet model: Wacom CTL 490DW-S

hawku, 2018-12-11:
@SgtCataclyst Are you running 32-bit Windows 10?
The latest driver zip file will provide 32-bit version of VMulti driver.

closed 2018-12-22 by hawku

Miguel2004, 2019-01-02:
Please could you help me with this problem of "devcon.exe failed".
I am using a small size ctl-471, and my operating system is windows 8.1 64-bit.

hawku, 2019-01-02:
@Miguel2004 I can't reproduce the problem, so it is very hard to help with that problem.
You could try to run `driver_vmulti\install_vmulti_64.bat` as an administrator.

Miguel2004, 2019-01-02:
I tried to open as administrator but the program opens and when I press enter to download it closes without downloading the driver.

hawku, 2019-01-02:
@Miguel2004 Maybe your anti-virus is blocking the devcon.exe?
Can you run the the devcon.exe in the command prompt?

Miguel2004, 2019-01-02:
I tried and it does not work. My anti-virus is disabled too.

hawku, 2019-01-02:
@Miguel2004 Try to unzip the driver to another folder. For example `C:\Temp\TabletDriver\`
If that doesn't help, you have to find a solution from google. For example:
https://social.msdn.microsoft.com/Forums/en-US/de483d5e-ffc9-41fd-b887-11cdfd874515/devcon-failed?forum=windowspro-audiodevelopment

Miguel2004, 2019-01-03:
[image]
This is the driver installation error

hawku, 2019-01-03:
@Miguel2004 I don't know why the installation fails and I can't reproduce the problem.
The install script and devcon.exe is from the XP-Pen drivers.
You could try this:
https://github.com/hawku/TabletDriver/wiki/FAQ#q-ive-installed-vmulti-driver-but-i-get-an-error-cant-open-vmulti-device

Miguel2004, 2019-01-03:
That did not work, either.

hawku, 2019-01-03:
@Miguel2004 The XP-Pen drivers doesn't work either?
I can't help with your problem. There is something wrong with your Windows installation that needs to be fixed.

Saicooh, 2024-01-01:
- open remove_vmulti_driver.bat
- then open cmd and put bcdedit /set TESTSIGNING ON
- reboot and try installing again...

at least that worked for me :)

semyuon, 2024-01-10:
> * open remove_vmulti_driver.bat
> 
>     * then open cmd and put bcdedit /set TESTSIGNING ON
> 
>     * reboot and try installing again...
> 
> 
> at least that worked for me :)

holy shit it actually worked