hawku/TabletDriver #230: Support for Genius MousePen i608x

KingDGrizzle, 2018-08-13:
Hello.

I've recently heard about this driver, and it's amazing.
But it doesn't seem to work with my tablet, which is why I'm making an issue here.

I can also help testing if you wanted me to!

<details><summary>Log Info</summary><p>

```
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x000C, Usage: 0x0001
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 3, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG] 
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG] 
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x0001, Usage: 0x0000
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 8, InputLen: 0, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG] 
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG]
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG] 
2018-08-13 11:23:40 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-08-13 11:23:40 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2018-08-13 11:23:40 [HIDDevice] [DEBUG] 
2018-08-13 11:23:41 [HIDDevice] [DEBUG] HID Device: Vendor: 'Genius' Product: 'MousePen i608X'
2018-08-13 11:23:41 [HIDDevice] [DEBUG]   Vendor Id: 0x0458, Product Id: 0x501A
2018-08-13 11:23:41 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2018-08-13 11:23:41 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
```

</p>
</details>

Drivers: https://transfer.sh/11onFG/ioTablet.zip

> I used transfer.sh because the drivers takes years to download from their website, so you won't have to wait minutes on end for it.

hawku, 2018-12-11:
@KingDGrizzle I tried to implement support for Genius i405X, but it seems to be impossible, so I don't think that i608X would work either.

closed 2018-12-11 by hawku

vladfrangu, 2019-04-11:
Dang, that sucks... Thanks for letting me know!