hawku/TabletDriver #240: Driver support for mac osx

Creamez, 2018-09-02:
i would really like it if you made driver support possible for all the mac osx players who player osu on there. And everybody that would want no input lag while using their drivers. And because right now mac does not really have great drivers they are not really as good as windows ones. So im just wondering if  you would ever take your time making support for mac osx

Banbeucmas, 2018-09-05:
Same as Linux. They got different way of handling the Monitor so I don’t expect it to be soon or fast

hawku, 2018-12-19:
@n0ts0bad Maybe you don't understand how much time it would take to convert the driver to Mac OS X?

I'd say it would take at least 200 hours of rewriting some parts of the driver and code a completely new user interface for Mac OS X.
After that it, if someone uses the driver on OS X, it would take off some development time to deal with the Mac OS problems.
I'd say that there is already enough problems with the Windows...

Maybe in the future when there isn't anything to develop or fix in the driver, I will modify the driver code so that it can be compiled in Linux.

After that it would be easier to someone to convert the code to Mac OS X.
I will not do the OS X conversion, because I have zero experience with the Apple products and I don't have a computer with a Mac OS X.

closed 2018-12-19 by hawku

Zen1thOrSomething, 2021-02-08:
ok so making mac osx drivers would take a while
well what else are we supposed to use? a driver with 2 second delay?
also, both linux and mac were built using unix, if you can do linux you can so mac.
also, unless you are the only developer who made the driver, can't you hire a mac developer?