hawku/TabletDriver #246: Deco 01 not working: Tablet Read Error

Dador, 2018-09-12:
Hi.
I have XP-Pen Deco 01 but the product id is different from the one in config file.

From HIDList:
```
2018-09-12 17:54:50 [HIDDevice] [DEBUG] HID Device: Vendor: 'UGTABLET' Product: 'DECO 01'
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   Vendor Id: 0x28BD, Product Id: 0x0902
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   Usage Page: 0xFF03, Usage: 0x0000
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 64, OutputLen: 0
2018-09-12 17:54:50 [HIDDevice] [DEBUG] 
2018-09-12 17:54:50 [HIDDevice] [DEBUG] HID Device: Vendor: 'UGTABLET' Product: 'DECO 01'
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   Vendor Id: 0x28BD, Product Id: 0x0902
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   Usage Page: 0xFF0A, Usage: 0x0001
2018-09-12 17:54:50 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 10, OutputLen: 8
```

I've tried to change product id in config to mine and TabletDriver started to recognize it. But now I get `Tablet Read Error`:
```
2018-09-12 17:54:50 [INFO] >> Start
2018-09-12 17:54:50 [INFO] TabletDriver started!
2018-09-12 17:54:50 [ERROR] Tablet Read Error!
2018-09-12 17:54:50 [STATUS] TABLET XP-Pen Deco 01
2018-09-12 17:54:50 [STATUS] HID 28BD 0902 FF0A 0001
2018-09-12 17:54:50 [STATUS] WIDTH 254.00000
2018-09-12 17:54:50 [STATUS] HEIGHT 158.75000
2018-09-12 17:54:50 [STATUS] MAX_X 25400
2018-09-12 17:54:50 [STATUS] MAX_Y 15875
2018-09-12 17:54:50 [STATUS] MAX_PRESSURE 8191
2018-09-12 17:54:50 [INFO] >> Log Off
```

How can I fix that error?

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.1.5

Tablet model: XP-Pen Deco 01

Dador, 2018-09-12:
Also the tablet itself is working (I can move the cursor etc), is that expected? Or does it means driver actually isn't used for the tablet?

hawku, 2018-12-11:
@Dador Try the latest version.
Your Deco 01 might be the new model that is supported in the latest TabletDriver version.

closed 2019-01-09 by hawku

Dador, 2019-01-09:
Missed previous message.

Thank you for your answer!