hawku/TabletDriver #255: Driver thinks tablet is a different tablet - Huionk58 but thinks it's a Huion osu!tablet

TuckieO, 2018-10-04:
I am using a Huion K58 but it thinks it's an osu! tablet and won't let me pick an area larger than it - which the tablet I am using has a much larger area than the Huion 420. It wouldn't work in a previous update at all, but randomly started working with this update so maybe add support for the Huion k58 in the future?
https://puu.sh/BFqhz.png
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.5

Tablet model: Huion K58

hawku, 2018-12-11:
@TuckieO Try the latest driver version and post the startuplog.txt here.

TuckieO, 2018-12-12:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2673136/startuplog.txt)

riley-badour, 2018-12-12:
Looks like this tablet has identifier strings but its hard to determine if we can define the tablet like this i dont see tablet string 201. Hawku i think we need that one to get this tablet . We do need a usblyzer file for us to get some information on this tablet. I can provide instuction on how to do this later i dont have acess to a computer right now.

hawku, 2018-12-12:
@TuckieO Open `config\tablet.cfg`, change `GetStrings 1 20` from the osu!tablet config to `GetStrings 1 250`, restart the driver and post an another startuplog.txt here.

If the `GetStrings 1 250` causes the tablet to disconnect, try `GetStrings 190 230`

TomSputz, 2019-02-13:
Played around with my tablet a bit to see what the driver could do:

| Official Driver | Open Driver | Effect | Strings |
|-|-|-|-|
| 🗙 | 🗙 | Small section of tablet tracks along bottom of screen - useless | 🗙 |
| Installed | 🗙 | Small section of tablet tracks along bottom of screen - useless | 🗙 |
| Running | 🗙 | Accurate mapping and customisation | 🗙 |
| 🗙 | Running | Tablet not detected - `Driver stopped!` | 🗙 |
| Installed | Running | Tracks accurately in 5cm area in top left of tablet | ✓ |

The strings found while the official driver is installed were:
```
1: "HV-LOGIC"
2: "Digitizer"
3-4: "HVAN"
5: "HUION"
6: "Pentablet"
7-99: "HVAN"
100: ""
101: " "
102: "HV-Logic TechnologyCorp."
103-109: "HVAN"
110: ""
111--120: "HVAN"
121: "M508"
122: ""
123: "HK On"
124: "HK Off"
125-250: "HVAN"
```

TomSputz, 2019-02-13:
Anyone coming across this thread because they want to use the driver with a K58, steps are currently as follows:

1. Download and install the official drivers: [Link](https://www.huion.com/download/type-1/?Keyword=k58)
2. Download and install this driver as normal
3. Replace config/tablet.cfg with [tablet.cfg](https://gist.github.com/TomSputz/d27c31bafbd893d7b15b9ea1e7f6a263)
4. Run `TabletDriverGUI.exe`

riley-badour, 2019-02-13:
Your tablet in reality is a 580 right thean we have to use string 121 to identify your driver

TuckieO, 2019-02-13:
Does the driver function properly now or is it just the top left 5cm like before?