hawku/TabletDriver #257: pressure sensitive not work in photoshop

you06, 2018-10-09:
Hi, I'm using this tablet driver for osu! game, thanks for your work. I have the problem when I'm trying to using it with photoshop that I can not enable pressure sensitive feature in photoshop. If this driver support pressure sensitive?

## Hardware/software information
Windows Version: 
win10 1803
Tablet Driver version: 
V0.1.5
Tablet model: 
Wacom CTL-471
Adobe Photoshop:
19.0 20170929.r.165 2017/09/29: 1138933  x64

hawku, 2018-10-09:
The driver only supports pressure sensitivity in Windows Ink mode.
The older WinTab interface that Photoshop uses is not supported.

you06, 2018-10-10:
@hawku thanks! Switch to Windows Ink mode works for me.
For cc 2017 and earlier version, create `PSUserConfig.txt` in photoshop setting directory ` C:\Users\[User Name]\AppData\Roaming\Adobe\Adobe Photoshop CC 201x\Adobe Photoshop CC 201x Settings\` with the following content
```
# Use WinTab
UseSystemStylus 0
```
#### delete `PSUserConfig.txt` if you are using cc 2018

closed 2018-10-10 by you06

you06, 2018-11-09:
@Aspiration-to-Stars Sorry I don't quite known with creating `PSUserConfig.txt` solution. I've tried it, but does not work with my Photoshop version.
Creating config file is an advice copy from [adobe forum](https://forums.adobe.com/thread/1331165). Maybe you can get more information from this post.