hawku/TabletDriver #258: Deco 01 not found.

satisnowflake, 2018-10-12:
Hello.
I have a XP-PEN Deco 01 tablet.
And your driver doesn't work with it.
Some people have the same issue, I think.
There's my startup *.txt file attached:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2472191/startuplog.txt)

## Hardware/software information
Windows Version: Windows 10 Professional x64.

Tablet Driver version: 
[image]

Tablet model: XP-PEN Deco 01.

hawku, 2018-10-12:
When did you purchase that tablet?
It seems like XP-Pen have made different DECO 01 models with different hardware/firmware.
Changing the `Tablet 0x28BD 0x0042 0xFF0A 0x0001` line to `Tablet 0x28BD 0x0902 0xFF0A 0x0001` should work if the tablet's firmware is similar to other XP-Pen models.

You could also try change the init report to: `InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00`

satisnowflake, 2018-10-12:
Tablet was bought in the July, 15th.
As you can see, I have already tried to change the «Tablet» line.
It's working very strangeful: it's always «Windows Inc» mode without a cursor.
Changing the «InitReport» line didn't fix it too.

closed 2018-11-27 by hawku