hawku/TabletDriver #259: Tablet area stuck in top left corner of tablet

Fatmancer, 2018-10-13:
Hello,

I think my tablet is technically unsupported, but everything else works other than moving the tablet area around, so i wanted to at least check if there is anything that can be done regarding this. I have a Huion H610PRO 8192.

The Tablet area GUI seems to think that my maximum available tablet area is only about 100mm by 56mm (see the first image below showing the full tablet area) set in the top left corner of the tablet itself. I can adjust the size of the tablet area properly, but cant move it past the offsets shown in the second picture since the GUI doesn't seem to think there is any tablet beyond this.

[image]

[image]

## Hardware/software information
Windows Version: Windows 10 Home V1803 Build 17134.345

Tablet Driver version: 0.1.5

Tablet model: Huion H610PRO 8192

BYSTACK, 2018-11-03:
You can try to edit osu!tablet's settings in config/tablet.cfg file to match your tablet

riley-badour, 2018-12-13:
you sem try using the latest version  and install the drivers again   after you do that can you post your  startuplog.txt here. with
you tablet is not  supported  but we can get your tablet to work i own the same tablet the new1060 8192 is a version of your 610  we just need that log  file and we can start to get your tablet to work

closed 2019-01-09 by hawku