hawku/TabletDriver #261: Compatibility with xp pen star g640s plsssssss

JetFury, 2018-10-15:
## Hardware/software information
Windows Version: windows 10

Tablet Driver version: the latest one as of Oct. 15 2018

Tablet model:  xp pen star g640s

nomq1, 2018-10-16:
try changing config file : tablet.cfg   
see #196 
([https://github.com/hawku/TabletDriver/pull/196/commits/62d126389a5e209b476efe4611475f2ee4e3af31](here more specifically )

Renarix, 2018-10-16:
did you change it or you need file?

Renarix, 2018-10-16:
i can change config file if you dont know how to do it

nomq1, 2018-10-16:
just go to the file at  [unzipped folder]/config/tablet.cfg 
and add to the end of the file the changes frodriguez96 suggested .
(or download the file from frodriguez96 repository [https://raw.githubusercontent.com/frodriguez96/TabletDriver/master/TabletDriverService/config/tablet.cfg](url) and replace it with your current one )

SilentNutMan, 2018-12-05:
> just go to the file at [unzipped folder]/config/tablet.cfg
> and add to the end of the file the changes frodriguez96 suggested .
> (or download the file from frodriguez96 repository [https://github.com/frodriguez96/TabletDriver/blob/master/TabletDriverService/config/tablet.cfg] and replace it with your current one )

It Is work! Thank you so match ^)