hawku/TabletDriver #266: Cursor jump after Windows 10 1809 update

bl4ckout31, 2018-10-21:
Since the Windows 10 1809 update, the driver goes into a "Driver stopped. Restarting" loop wich causes the cursor to teleport to the upper left corner of the screen everytime the driver restart **when the tablet is not connected**. 
Everything works fine as long as the tablet is connected.

The cursor jump wasn't happening before the 1809 update, but I don't know about the driver restart loop.
I removed and reinstalled the driver to no effect and the tablet was disconnected when I did the update.

Here is the [startuplog.txt](https://github.com/hawku/TabletDriver/files/2499066/startuplog.txt) in case there is something relevant.

## Hardware/software information
Windows Version: Windows 10 version 1809

Tablet Driver version: V0.1.5

Tablet model: Wacom CTL-480

wojexe, 2018-10-26:
I had this issue on Windows 10 Version 1803 and still have it on Version 1809.

bl4ckout31, 2018-12-11:
I comfirm that https://github.com/hawku/TabletDriver/releases/tag/v0.2.1 has fixed my issue.
Closing.

closed 2018-12-11 by bl4ckout31