hawku/TabletDriver #269: Driver to draw

shibou3, 2018-10-24:
Is it possible to use this driver to draw? I have a CTL-490 and in the past I could use to draw (pressure thing) by using "windows ink" option but when I try to do it now, it doesn't work?

Hummart, 2018-10-29:
v0.0.15:

        _Removed ClickPressure workaround from Wacom 490. You can re-enable that by adding 
        ClickPressure 500 to Commands tab._

Aspiration-to-Stars, 2018-11-08:
If I add "ClickPressure 500" to the Commands tab (I also tried to execute in the the console of the driver) the pen stops clicking at all (so no drawing, only movements).

hawku, 2018-12-13:
@shibou3 Does you drawing application support Window Ink?
TabletDriver does not support Wacom's old wintab32.dll, so some applications will not have pen pressure if they only use wintab pen pressure.

shibou3, 2018-12-13:
> @shibou3 Does you drawing application support Window Ink?
> TabletDriver does not support Wacom's old wintab32.dll, so some applications will not have pen pressure if they only use wintab pen pressure.

I use PS and Paint Tool Sai, I believe in the past Paint Tool Sai used to support it but now it doesn't work

hawku, 2018-12-13:
Photoshop might need some special settings: https://forums.adobe.com/thread/1500227
It seems that Paint Tool Sai doesn't have Windows Ink support.

shibou3, 2018-12-13:
Tbh I redownload it after a long time without using it (because of wacom drivers that I use to draw) and I notice that now the option works with CTL490, so I don't really "need" the ink support since I can turn on/off the driver whenever I want, thanks for that.

closed 2018-12-13 by shibou3