hawku/TabletDriver #274: Full tablet area

sevenbbt, 2018-11-01:
I have unistalled pentablet driver before running v_multi
I have also changed my tablet area but somehow it still uses up my full tablet area, just.. the speed was different, it did not disable any of the unwanted tablet area.
I have applied, saved, and restarted the driver, still nothing. 

## Hardware/software information
Windows Version: 10 Home

Tablet Driver version: v0.1.5

Tablet model: XP-Pen G430s
[image]

BYSTACK, 2018-11-03:
G430s is not in the supported tablets list. though there is PR #196

hawku, 2018-12-10:
@sevenbbt Please, check the issue #301

closed 2019-01-09 by hawku