hawku/TabletDriver #280: CTH-480 (wacom drivers installed)

ItsDavDav, 2018-11-13:
I recently switched from my CTL-472 to the CTH-480. I don't have any driver except for this one and it worked fine for my 472 but somehow it doesn't for my 480. It says CTH-480 (wacom drivers installed) and it won't use the settings of this driver. I've tried almost everything and cant find a solution.

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.5

Tablet model: CTH-480

hawku, 2018-11-21:
Windows 10 may have automatically installed Wacom drivers for the CTH-480.
Check this: https://github.com/hawku/TabletDriver/issues/1#issuecomment-372568808

ItsDavDav, 2018-11-21:
> Windows 10 may have automatically installed Wacom drivers for the CTH-480.
> Check this: [#1 (comment)](https://github.com/hawku/TabletDriver/issues/1#issuecomment-372568808)

I looked in my device manager but there are no wacom drivers at all. do you maybe have any other idea as to what the problem would be?

ricardobrug, 2018-11-29:
go to start, start typing uninstall and look for uninstall programs in control panel. Look for anything with the word tablet that has wacom as the dev/creator. I had the same issue.

ItsDavDav, 2018-11-29:
After some fiddling I finally managed to get it to work. There were infact some wacom drivers installed but they were hidden so I couldn't find them normally. 

Its all good now though!

closed 2018-12-08 by hawku