hawku/TabletDriver #281: ERROR I Can't save my custom config

Iscarlat9, 2018-11-13:
## Hardware/software information
Windows Version: Windows 10 64Bits

Tablet Driver version: latest

Tablet model: CTH-472
[image]

I can't save my own config anymore idk why, now every time that i open the driver it starts with the default config

Kiewaii, 2018-12-15:
> Well, can you create files in that folder?

Please help me, i cant use the driver because of that. How can i change that? How can I give myself the permission?

hawku, 2018-12-15:
@Kiewaii How did you unzip the driver to a folder that you don't have permissions to write?
Maybe try another folder.

closed 2018-12-22 by hawku

Powerfulwho1783, 2020-03-03:
Make sure that you put the folder or zip file on your desktop then open