hawku/TabletDriver #284: Clarification for the left handed mode

EndJoseph, 2018-11-17:
Hey, I don't know if I'm missing the point but I can't seem to understand why inverting the area serves as the left handed mode. Like, am I supposed to rotate the tablet itself or am I missing something else?

## Hardware/software information
Windows Version: win10

Tablet Driver version: 0.1.5

Tablet model: CTH-480

Joehuu, 2018-11-17:
`Left handed (Inverted)` is just a setting for convenience, so you do not have to manually change the rotation to 180 degrees.

EXtremeExploit, 2018-11-18:
i'm actually left handed and i never checked that option or rotated the area. I just use it as normal

Joehuu, 2018-11-18:
It depends what tablet you have. Some designs require rotating for left handed usage.

hawku, 2018-11-29:
I am left handed too and I only use this option with the XP-PEN G430.
It's also useful when you want have the USB cable in different side of the tablet.

closed 2018-11-29 by hawku