hawku/TabletDriver #285: G640 & antichatter

JohnathanKeen, 2018-11-21:
Hi,

When I enable antichatter while using my G640, it doesn't seem to be doing anything and as soon as I start using mouse, the cursor position is locked to the last tablet position. 

This does not happen if I enable antichatter with my CTL-480,

I have installed both vmulti driver and huion driver just in-case, still nothing. Happens purely with anti-chatter.

[image]

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.1.5.5 Devocub Edition

Tablet model: XP-PEN G640

hawku, 2018-11-21:
@JohnathanKeen I think that the XP-PEN G640 doesn't need any antichatter or smoothing from the drivers. That tablet does the pen position smoothing and noise suppression on the hardware. Hardware smoothing causes already enough of latency, so I wouldn't recommend using any additional filters.

JohnathanKeen, 2018-11-21:
> @JohnathanKeen I think that the XP-PEN G640 doesn't need any antichatter or smoothing from the drivers. That tablet does the pen position smoothing and noise suppression on the hardware. Hardware smoothing causes already enough of latency, so I wouldn't recommend using any additional filters.

I agree with you, just thought I'd share since it seems to be an actual bug. Up to @Devocub if he's willing to investigate it or not.

Thanks for the replies.

closed 2018-12-11 by hawku