hawku/TabletDriver #288: halp

Eranuwu, 2018-11-23:
VCRUNTIME140.dll is missing

https://i.imgur.com/OaZiyyV.png

## Hardware/software information
Windows Version: Windows 10 Pro 64bit 

Tablet Driver version: v 0.1.5 

Tablet model: CTH 470

hawku, 2018-11-23:
You need to install the C++ runtime library.
https://github.com/hawku/TabletDriver#installation
https://aka.ms/vs/15/release/vc_redist.x86.exe

Eranuwu, 2018-11-23:
I HAVE IT DOEEEEE

RisedSky, 2018-11-24:
http://www.telecharger-dll.fr/dll-vcruntime140.dll.html

Put the dll in `C:\Windows\System32`

If you use 64bits put **too** in `C:\Windows\SysWOW64\`

hawku, 2018-11-24:
@Eranuwu I just checked the picture you provided: https://i.imgur.com/OaZiyyV.png
That error isn't for this driver... That is an error for iCUE.exe, which is Corsair Utility Engine...

closed 2018-11-24 by hawku