hawku/TabletDriver #290: Unable to read a backup file

podww, 2018-11-27:
https://extrathicc.s-ul.eu/S6TMkpXW is the backup file.

## Hardware/software information
Windows Version: Windows 10 home

Tablet Driver version: v0.15

Tablet model: Huion 420

hawku, 2018-11-27:
@podww Your tablet is Huion 420, but that backup file is for Wacom tablets.
Which Wacom tablet that backup file is for?
I haven't seen that file format before, so that means that the file format isn't supported.

podww, 2018-11-27:
@hawku It is my backup file from the Wacom CTH 460.

hawku, 2018-11-29:
@podww  Are these the correct tablet area values in that file:
```
InputOriginX 1742
InputOriginY 1265
InputOriginZ 0
InputExtentX 3726
InputExtentY 2075
InputExtentZ 0
```

podww, 2018-11-29:
@hawku I believe so, but I couldn't tell you as I don't own that tablet anymore.