hawku/TabletDriver #291: Tablet tapping not working properly

KSN24, 2018-11-30:
When the input of my pen tip is enabled, it just keeps randomly activating (pressing mouse1 / left mouse button) even if it is not supposed to, for example when I'm hovering, also actual pressing sometimes doesn't work. 

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.5

Tablet model: CTH-480

hawku, 2018-12-01:
@KSN24 Does that happen with the official Wacom drivers?

KSN24, 2018-12-01:
@hawku no, that's why I opened an issue. I considered going back to wacom drivers so I can use it again, but I decided to check if you were able to fix my issue instead.

hawku, 2018-12-22:
@KSN24 Maybe the pen nib is stuck so it will apply little bit of pressure to pressure sensor?
Try another pen nib or reinsert the current nib.

You can also try to add this command to commands tab: `ClickPressure 100`
That will set the pen tip click pressure limit to about 10% of maximum pressure.

KSN24, 2018-12-22:
pennip wasn't actually stuck, but reinserted it and now it seems to work? don't get it to be honest.

eitherway, thanks for your advice, it now works again!

closed 2018-12-22 by KSN24