hawku/TabletDriver #292: Driver settings not working as intended

JaViLuMa, 2018-12-01:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version:  0.1.5

Tablet model:  Huion H420

Drivers sometimes (like 3-5 times a day) just stop working, cursor doesn't move and if it does move it is inverted (I am left handed) from the way it should be moving.

Restarted drivers like 20+ times and the issue still happens.

hawku, 2018-12-01:
@JaViLuMa What does it say in the console tab when the tablet doesn't work correctly?
Upload the `startuplog.txt` from the drivers folder to here when the problem occurs next time.

JaViLuMa, 2018-12-01:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2636385/startuplog.txt)

Here is the log. The cursor goes to the top left of the screen in osu! and the cursor is inverted.

hawku, 2018-12-11:
@JaViLuMa Does that happen with the latest version (0.2.1)?

JaViLuMa, 2018-12-15:
@hawku I don't seem to have any issues this last 2 weeks.

closed 2018-12-22 by hawku