hawku/TabletDriver #298: Cursor regularly glitches

Milessweet418, 2018-12-09:
Cursor will jump to a different location for a very brief period, before switching back, many times a second.
Debug and startup log attached, just in case you need it.
[debug_2018-12-09_13_52_35.txt](https://github.com/hawku/TabletDriver/files/2660205/debug_2018-12-09_13_52_35.txt)
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2660206/startuplog.txt)

## Hardware/software information
Windows Version: Win10 64bit

Tablet Driver version: 0.2

Tablet model: XP-Pen G640

hawku, 2018-12-09:
@Milessweet418 You are using Windows Ink mode. That will allow the normal mouse to have its own position.

If you move pen and mouse at the same time, the cursor will jump between mouse and pen position.

Milessweet418, 2018-12-09:
It happens on all modes.

hawku, 2018-12-09:
I generated an image from the debug log file:
[image]

To me that looks normal, just some XP-PEN specific low speed jumps.
Are those small jumps the problem?
Are you using extra small tablet area?

hawku, 2018-12-09:
@Milessweet418 Are you sure that you are not running XP-Pen driver processes PentabletService.exe or PenTablet.exe in the background?

Milessweet418, 2018-12-09:
PenTablet is not running in the background. It's quite annoying that the problem isn't showing up on the image. I'll create a quick video displaying the problem and get it to you ASAP.

Milessweet418, 2018-12-09:
Never mind. I forgot to check for PentabletService. Problem resolved, thanks for the help, and sorry for being a bit stupid on my end.

closed 2018-12-11 by hawku