hawku/TabletDriver #299: The Application was Unable to Start Correctly (0xc000007b)

gaseet, 2018-12-10:
## Hardware/software information
Windows Version: 8.1

Tablet Driver version: 0.2

Tablet model: Wacom CTH-480

Already tried everything I could find on this error, including things on past posts about the same error, nothing fixed it, before, I was getting an error about the MSCVP140.dll missing, but I fixed it by copying and pasting the dll from the 64-bit system file into the 32-bit system file (System32-SysWOW64). And after that, I keep getting this error.

(https://user-images.githubusercontent.com/45747276/49711418-50d6a080-fc26-11e8-8b57-645fe978ae7e.PNG)

hawku, 2018-12-10:
@gaseet So you are trying to start a 32-bit program that needs 32-bit msvcp140.dll, but you have changed that file to a 64-bit version???

The driver installation instructions tells to install https://aka.ms/vs/15/release/vc_redist.x86.exe
That installer should provide the necessary 32-bit versions of Visual C++ runtime libraries, including msvcp140.dll

gaseet, 2018-12-10:
I have tried installing the Visual C++ that is provides in the instructions, but the installer freezes everytime.

gaseet, 2018-12-10:
Managed to fix it, thanks for the quick reply though.

closed 2018-12-11 by hawku

Avoreas, 2021-01-09:
> Мне удалось это исправить, хотя Спасибо за быстрый ответ.

Hi, how did you manage to solve this problem?