hawku/TabletDriver #301: XP-Pen G430S and G540 support. Testers needed!

hawku, 2018-12-10:
I need help from people that own a XP-Pen G430S or G540 tablet.
Currently there is no way to determine if the tablet is G430, G430S or G540, so I need more data to differentiate those tablets from each other.

**Instructions:**
1. Download the latest TabletDriver version (v0.2.1 or greater).

2. Edit `config\user.cfg` file to this:

```
# Close startuplog.txt
Log off

# Start logging to xppen.txt
Log XPPenStrings.txt

# Close tablet connection
CloseTablet

# Open new tablet connection
HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
GetStrings 1 4
GetStrings 102
GetStrings 110
GetStrings 131
GetStrings 168
Log off
Wait 999999
```

3. Connect the tablet to the PC and start the TabletDriverGUI.
That will create a `XPPenStrings.txt` to the TabletDriver folder.

4. Post a comment here that includes contents of that file.

The file should contain something like this:
```
[2018-12-10 13:18:47.407] Log file 'XPPenStrings.txt' opened.
[2018-12-10 13:18:47.407] >> # Close tablet connection
[2018-12-10 13:18:47.407] >> CloseTablet
[2018-12-10 13:18:47.407] Tablet closed!
[2018-12-10 13:18:47.407] >> # Open new tablet connection
[2018-12-10 13:18:47.407] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2018-12-10 13:18:47.409] Tablet found!
[2018-12-10 13:18:47.409] >> GetStrings 1 4
[2018-12-10 13:18:47.409] Requesting string ids from 1 to 4:
[2018-12-10 13:18:47.409]   String (1) = 'UGTABLET'
[2018-12-10 13:18:47.410]   String (2) = 'TABLET G3 4x3'
[2018-12-10 13:18:47.410]   String (3) = 'V2.0 20170423'
[2018-12-10 13:18:47.410]   String (4) = 'V2.0 20170423'
[2018-12-10 13:18:47.410] >> GetStrings 102
[2018-12-10 13:18:47.410] Requesting string id 102:
[2018-12-10 13:18:47.411]   String (102) = 'UGTizer Technology Corp.'
[2018-12-10 13:18:47.411] >> GetStrings 110
[2018-12-10 13:18:47.411] Requesting string id 110:
[2018-12-10 13:18:47.411]   String (110) = '0'
[2018-12-10 13:18:47.411] >> GetStrings 131
[2018-12-10 13:18:47.411] Requesting string id 131:
[2018-12-10 13:18:47.412]   String (131) = 'V2.0 20170423'
[2018-12-10 13:18:47.412] >> GetStrings 168
[2018-12-10 13:18:47.412] Requesting string id 168:
[2018-12-10 13:18:47.412]   String (168) = 'V2.0 20170423'
[2018-12-10 13:18:47.412] >> Log off
```
5. Remove all lines from the `config\user.cfg` to get the driver back to normal operation.

riley-badour, 2018-12-10:
[2018-12-10 07:09:10.741] Log file 'XPPenStrings.txt' opened.
[2018-12-10 07:09:10.741] >> # Close tablet connection
[2018-12-10 07:09:10.741] >> CloseTablet
[2018-12-10 07:09:10.741] Tablet closed!
[2018-12-10 07:09:10.741] >> # Open new tablet connection
[2018-12-10 07:09:10.741] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2018-12-10 07:09:10.744] Tablet found!
[2018-12-10 07:09:10.744] >> GetStrings 1 4
[2018-12-10 07:09:10.744] Requesting string ids from 1 to 4:
[2018-12-10 07:09:10.746]   String (1) = 'XP-PEN'
[2018-12-10 07:09:10.749]   String (2) = 'G430S'
[2018-12-10 07:09:10.752]   String (3) = 'V2.0 20171204'
[2018-12-10 07:09:10.755]   String (4) = 'V2.0 20171204'
[2018-12-10 07:09:10.755] >> GetStrings 102
[2018-12-10 07:09:10.755] Requesting string id 102:
[2018-12-10 07:09:10.758]   String (102) = 'UGTizer Technology Corp.'
[2018-12-10 07:09:10.758] >> GetStrings 110
[2018-12-10 07:09:10.758] Requesting string id 110:
[2018-12-10 07:09:10.761]   String (110) = '0'
[2018-12-10 07:09:10.761] >> GetStrings 131
[2018-12-10 07:09:10.761] Requesting string id 131:
[2018-12-10 07:09:10.764]   String (131) = 'V2.0 20171204'
[2018-12-10 07:09:10.764] >> GetStrings 168
[2018-12-10 07:09:10.764] Requesting string id 168:
[2018-12-10 07:09:10.767]   String (168) = 'V2.0 20171204'
[2018-12-10 07:09:10.767] >> Log off

this is great string 2 give you what you need

hawku, 2018-12-10:
@riley-badour Thanks!

Add this to the `config\tablet.cfg`:
```
#
# XP-Pen G430S
#
Tablet 0x28BD 0x0075 0xFF0A 0x0001
CheckString 2 "G430S"
Name "XP-Pen G430S"
ReportId 0x02
ReportLength 8
DetectMask 0x80
MaxX 20320
MaxY 15240
MaxPressure 8191
Width 101.6
Height 76.2
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00
```
Also, edit the G430 configuration from this:
```
#
# XP-Pen G430
#
HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
Name "XP-Pen G430"
```
to
```
#
# XP-Pen G430
#
HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
CheckString 2 "TABLET G3"
Name "XP-Pen G430"
```

riley-badour, 2018-12-10:
you forgot the HID infront Tablet  but it does work my  g430s is still garbage  at 10mm x 10mm compared to my friends any way Great work  works edge to edge

Cliche45, 2018-12-10:
I've been trying the said fix and it doesn't seem to work for some reason. How to fix ;-; 
[image] 
[image]

hawku, 2018-12-10:
@Cliche45 Sorry, I had an error on the G430S config.
Change `Tablet 0x28BD...` to `HIDTablet 0x28BD...`.
I have now edited the comment.

sssata, 2018-12-12:
Here's my XPPenStrings.txt

```
[2018-12-11 19:21:12.135] Log file 'XPPenStrings.txt' opened.
[2018-12-11 19:21:12.135] >> # Close tablet connection
[2018-12-11 19:21:12.135] >> CloseTablet
[2018-12-11 19:21:12.135] Tablet closed!
[2018-12-11 19:21:12.135] >> # Open new tablet connection
[2018-12-11 19:21:12.135] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2018-12-11 19:21:12.136] Tablet found!
[2018-12-11 19:21:12.137] >> GetStrings 1 4
[2018-12-11 19:21:12.137] Requesting string ids from 1 to 4:
[2018-12-11 19:21:12.137]   String (1) = 'XP-PEN'
[2018-12-11 19:21:12.137]   String (2) = 'G430S'
[2018-12-11 19:21:12.138]   String (3) = 'V2.0 20171204'
[2018-12-11 19:21:12.138]   String (4) = 'V2.0 20171204'
[2018-12-11 19:21:12.138] >> GetStrings 102
[2018-12-11 19:21:12.138] Requesting string id 102:
[2018-12-11 19:21:12.139]   String (102) = 'UGTizer Technology Corp.'
[2018-12-11 19:21:12.139] >> GetStrings 110
[2018-12-11 19:21:12.139] Requesting string id 110:
[2018-12-11 19:21:12.140]   String (110) = '0'
[2018-12-11 19:21:12.140] >> GetStrings 131
[2018-12-11 19:21:12.140] Requesting string id 131:
[2018-12-11 19:21:12.140]   String (131) = 'V2.0 20171204'
[2018-12-11 19:21:12.140] >> GetStrings 168
[2018-12-11 19:21:12.140] Requesting string id 168:
[2018-12-11 19:21:12.140]   String (168) = 'V2.0 20171204'
[2018-12-11 19:21:12.140] >> Log off
```

Seems to be the same as above posts.

I only have a G430S to test, but your tablet.cfg edit works perfect for me, thanks.

Arcandael, 2019-02-26:
[2019-02-26 19:24:37.952] Log file 'XPPenStrings.txt' opened.
[2019-02-26 19:24:37.952] >> # Close tablet connection
[2019-02-26 19:24:37.952] >> CloseTablet
[2019-02-26 19:24:37.952] Tablet closed!
[2019-02-26 19:24:37.952] >> # Open new tablet connection
[2019-02-26 19:24:37.952] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2019-02-26 19:24:37.955] Tablet found! Manufacturer='UGTABLET', Product='TABLET G3 5x4', Serial=''
[2019-02-26 19:24:37.955] >> GetStrings 1 4
[2019-02-26 19:24:37.955] Requesting string ids from 1 to 4:
[2019-02-26 19:24:37.956]   String (1) = 'UGTABLET'
[2019-02-26 19:24:37.956]   String (2) = 'TABLET G3 5x4'
[2019-02-26 19:24:37.957]   String (3) = 'V2.0 20170711'
[2019-02-26 19:24:37.958]   String (4) = 'V2.0 20170711'
[2019-02-26 19:24:37.958] >> GetStrings 102
[2019-02-26 19:24:37.958] Requesting string id 102:
[2019-02-26 19:24:37.958]   String (102) = 'UGTizer Technology Corp.'
[2019-02-26 19:24:37.958] >> GetStrings 110
[2019-02-26 19:24:37.958] Requesting string id 110:
[2019-02-26 19:24:37.959]   String (110) = '0'
[2019-02-26 19:24:37.959] >> GetStrings 131
[2019-02-26 19:24:37.959] Requesting string id 131:
[2019-02-26 19:24:37.959]   String (131) = 'V2.0 20170711'
[2019-02-26 19:24:37.959] >> GetStrings 168
[2019-02-26 19:24:37.959] Requesting string id 168:
[2019-02-26 19:24:37.960]   String (168) = 'V2.0 20170711'
[2019-02-26 19:24:37.960] >> Log off

This is the string given from the XP-Pen G540

Deviationlark, 2020-02-14:
[2020-02-14 20:21:09.649] Log file 'XPPenStrings.txt' opened.
[2020-02-14 20:21:09.649] >> # Close tablet connection
[2020-02-14 20:21:09.649] >> CloseTablet
[2020-02-14 20:21:09.649] Tablet is not open!
[2020-02-14 20:21:09.649] >> # Open new tablet connection
[2020-02-14 20:21:09.649] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2020-02-14 20:21:09.650] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0075 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
[2020-02-14 20:21:09.650] >> GetStrings 1 4
[2020-02-14 20:21:09.650] >> GetStrings 102
[2020-02-14 20:21:09.650] >> GetStrings 110
[2020-02-14 20:21:09.650] >> GetStrings 131
[2020-02-14 20:21:09.650] >> GetStrings 168
[2020-02-14 20:21:09.650] >> Log off

this is the string i got from xp-pen g430S and the tablet isnt working @hawku can you help?
i edited the cfg to what you said and its still not getting recognized for some reason...

xXNik0DarkXx, 2020-06-22:
XPPenStrings.txt from XP-Pen G540
[2020-06-22 08:34:28.196] Log file 'XPPenStrings.txt' opened.
[2020-06-22 08:34:28.196] >> # Close tablet connection
[2020-06-22 08:34:28.196] >> CloseTablet
[2020-06-22 08:34:28.196] Tablet closed!
[2020-06-22 08:34:28.196] >> # Open new tablet connection
[2020-06-22 08:34:28.196] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2020-06-22 08:34:28.211] Tablet found! Manufacturer='UGTABLET', Product='TABLET G3 5x4', Serial=''
[2020-06-22 08:34:28.211] >> GetStrings 1 4
[2020-06-22 08:34:28.211] Requesting string ids from 1 to 4:
[2020-06-22 08:34:28.211]   String (1) = 'UGTABLET'
[2020-06-22 08:34:28.211]   String (2) = 'TABLET G3 5x4'
[2020-06-22 08:34:28.211]   String (3) = 'U608g'
[2020-06-22 08:34:28.211]   String (4) = 'U608g'
[2020-06-22 08:34:28.211] >> GetStrings 102
[2020-06-22 08:34:28.211] Requesting string id 102:
[2020-06-22 08:34:28.211]   String (102) = 'UGTizer Technology Corp.'
[2020-06-22 08:34:28.211] >> GetStrings 110
[2020-06-22 08:34:28.211] Requesting string id 110:
[2020-06-22 08:34:28.211]   String (110) = '0'
[2020-06-22 08:34:28.211] >> GetStrings 131
[2020-06-22 08:34:28.211] Requesting string id 131:
[2020-06-22 08:34:28.211]   String (131) = 'U608g'
[2020-06-22 08:34:28.211] >> GetStrings 168
[2020-06-22 08:34:28.211] Requesting string id 168:
[2020-06-22 08:34:28.211]   String (168) = 'U608g'
[2020-06-22 08:34:28.211] >> Log off

@hawku

NIMKAOriginal, 2020-09-04:
```
2020-09-05 00:55:15 [INFO] Log file 'XPPenStrings.txt' opened.
2020-09-05 00:55:15 [INFO] >> # Close tablet connection
2020-09-05 00:55:15 [INFO] >> CloseTablet
2020-09-05 00:55:15 [WARNING] Unknown command: CloseTablet
2020-09-05 00:55:15 [INFO] >> # Open new tablet connection
2020-09-05 00:55:15 [INFO] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
2020-09-05 00:55:15 [WARNING] Unknown command: HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
2020-09-05 00:55:15 [INFO] >> GetStrings 1 4
2020-09-05 00:55:15 [WARNING] Unknown command: GetStrings 1 4
2020-09-05 00:55:15 [INFO] >> GetStrings 102
2020-09-05 00:55:15 [WARNING] Unknown command: GetStrings 102
2020-09-05 00:55:15 [INFO] >> GetStrings 110
2020-09-05 00:55:15 [WARNING] Unknown command: GetStrings 110
2020-09-05 00:55:15 [INFO] >> GetStrings 131
2020-09-05 00:55:15 [WARNING] Unknown command: GetStrings 131
2020-09-05 00:55:15 [INFO] >> GetStrings 168
2020-09-05 00:55:15 [WARNING] Unknown command: GetStrings 168
2020-09-05 00:55:15 [INFO] >> Log off

```

GenericSkrub, 2020-10-30:
[2020-10-29 20:42:10.871] Log file 'XPPenStrings.txt' opened.
[2020-10-29 20:42:10.871] >> # Close tablet connection
[2020-10-29 20:42:10.871] >> CloseTablet
[2020-10-29 20:42:10.871] Tablet is not open!
[2020-10-29 20:42:10.871] >> # Open new tablet connection
[2020-10-29 20:42:10.871] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2020-10-29 20:42:10.874] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0075 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
[2020-10-29 20:42:10.874] >> GetStrings 1 4
[2020-10-29 20:42:10.874] >> GetStrings 102
[2020-10-29 20:42:10.874] >> GetStrings 110
[2020-10-29 20:42:10.882] >> GetStrings 131
[2020-10-29 20:42:10.882] >> GetStrings 168
[2020-10-29 20:42:10.882] >> Log off

RivalsChoco, 2020-12-18:
[2020-12-18 17:26:48.966] Log file 'XPPenStrings.txt' opened.
[2020-12-18 17:26:48.966] >> # Close tablet connection
[2020-12-18 17:26:48.966] >> CloseTablet
[2020-12-18 17:26:48.967] Tablet is not open!
[2020-12-18 17:26:48.967] >> # Open new tablet connection
[2020-12-18 17:26:48.967] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
[2020-12-18 17:26:48.968] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0075 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
[2020-12-18 17:26:48.968] >> GetStrings 1 4
[2020-12-18 17:26:48.968] >> GetStrings 102
[2020-12-18 17:26:48.968] >> GetStrings 110
[2020-12-18 17:26:48.968] >> GetStrings 131
[2020-12-18 17:26:48.968] >> GetStrings 168
[2020-12-18 17:26:48.968] >> Log off