hawku/TabletDriver #302: Huion 1060 Plus support needs extra work

riley-badour, 2018-12-10:
## Hardware/software information
Windows Version: 10 1803

Tablet Driver version: 2.1

Tablet model: huion 1060 plus 8192

#
# Huion New 1060 Plus 8192  
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}" 
Name "Huion New 1060 Plus 8192 "
CheckString 201 "HUION_T174"
ReportId 0x08
ReportLength 12
DetectMask 0x80
IgnoreMask 0x60 #
MaxX 50800
MaxY 31750
MaxPressure 8191
Width  254
Height 158.75
InitStrings 100 200

this config works but this tablet has 2 more variables
after 80  we have x and y  then we have what is the outer pieces of the tablet  they  use x and y and whats seem to be reverse x and y code from the right to the left 
im guessing they ran out of address space ??
 
reverse x goes from right to left 00 to 7f
reverse y go from up top to bottom 00 to 7c 
theses values do go up and down during regular x and y 
im guessing they are only needed at max value

https://drive.google.com/file/d/1zxEfhhgZgC-SRgNDxiXSfc0qckkk_mGA/view?usp=sharing

usblyzer file with me moving my pen into each Corner of my tablet top left to top right  then bottom right to left
if this tablet is to hard to integrate forget about the extra .5cm to the right and 10 mm to the bottom and the config up top works fine

hawku, 2018-12-10:
@riley-badour I can't find any problems from your USBlyzer capture file.

There are just 12 bytes long HID reports with 8 bytes of normal tablet data and rest is empty.
The capture also shows X, Y and pressure values at the limits of 50800, 31750 and 8191.
Does the official Huion driver allow to use those extra 5 and 10 millimeters?

**Report data bytes:**
1. Report Id (0x08)
2. Buttons (0x80 = In range, Bit 1 = button 1, etc.)
3. Low X byte.
4. High X byte (X = Low + High * 256).
5. Low Y byte.
6. High Y byte (Y = Low + High * 256).
7. Low pressure byte.
8. High pressure byte (Pressure = Low + High * 256).
9. Empty
10. Empty
11. Empty
12. Empty

hawku, 2018-12-10:
@riley-badour I just checked my Huion H640P and it seems to miss about 5 millimeters from the left, so the tablet width doesn't match with the 5080 LPI value. 

Correct tablet area width for my H640P is 155 mm, not 160 mm.
Height does match with the 100 mm value.
I don't know if that is just a quality control problem or are every tablet made that way???

You can test the accuracy of tablet dimensions by using `Measure` console command.
- Run `Measure 2` and click two points with a pen. A click is registered when the pen pressure is released.
- Measure the distance between those clicks with a ruler and check it matches with the value given by `Measure` command.
- Keep the pen at the same angle when measuring distance, because pen angle difference will affect results.
- Do the measurements in the middle of the tablet to make sure that the tablet borders doesn't affect the results.
- Measure for X and Y axis separately.
- The results should be within a millimeter from the ruler value.

riley-badour, 2018-12-10:
[image]
some design of mine they must misprinted on the label for most of them  the height is perfect   but not to the boarders width is not to spec its short by a 1/4 of a inch from the spec  
original or yours same boarder area as shown with photo did a measure  human error was off by 0.05mm other than that this tablet is great a 285hz  refresh on it from osu  sd card reader kills that
who needs that anyway its off by .25 of a inch in width i sadly im short on tiny rulers this year to get you mm  im just wondering what do i have to do now with width vs lpi and that

hawku, 2018-12-10:
@riley-badour Set the tablet width to 248 mm and test the X axis again with the `Measure` command.

I don't know if every Huion New 1060 Plus model have that same problem, so maybe 250 mm width would be a good middle ground.

riley-badour, 2018-12-10:
250 is great  hits at at the max point measurement was off by 0.01mm  the other new 1060 plus even if they did have a misprint or not its still the same active area as determined by the numbers this tablet is also a copy of m106k they must be getting them cheap works edge to edge and it maxes out on width tablet is supported but im guessing we need more testers  there was a few older 1060s issues on here  and one plus version

hawku, 2018-12-10:
@riley-badour How bad is the hardware smoothing on 1060 plus?
For me the Huion H640P is almost unusable in osu! without anti-smoothing filter.
Does the recommended H640P's anti-smoothing filter values of Shape=0.7 and Compensation=8.0 work on 1060 too?

hawku, 2018-12-11:
Yeah the anti-smoothing doesn't work that great on XP-Pen tablets.
XP-Pen something weird at low speeds, so it is almost impossible to predict what the next position or speed will be.
It's like they are stopping the pen position when the speed is low enough and start moving again when the pen is far enough from that stop position.
It may be hardware resolution or firmware problem, but it is hard to tell.

&nbsp;

Does the 1060 with 0.7/8.0 anti-smoothing settings look like this when drawing circles with 80x45mm tablet area:
[image]

Or like this:
[image]

If it's like the first one, then the tablet have zero hardware smoothing.

&nbsp;

I checked your 1060 debug log and ran the data through anti-smoothing filter.
There is something wrong with tablet data. There are lots of perfect artificial straight lines, so the anti-smoothing filter can't predict the next position correctly.
When the cursor moves faster, the data looks normal and anti-smoothing filter can do it's job.

Your tablet area is insanely small, so that will cause problems with tablets that create those artificial straight lines.
I recommend that you learn to use larger area, because almost every tablet have problems with those kind of tablet areas.
Normal pen signal noise can be about 0.1 mm when dragging and can go up to 0.5 mm when hovering.

I tested the H640P with 10x5.63mm tablet area and it also seems to produce those artificial lines.

riley-badour, 2018-12-11:
tested it twice the second time as with both 1060s it glitched out  when slow with same area and anti smoothing  80x45 with usblyzer file

and image and yes i do push tablets to the smaller limits  a bit much can do a 20 by 22 just prefer smaller area

https://drive.google.com/file/d/1lm1PKlatZ4I5N4R0nHjq-Mf-d-bM1i5m/view?usp=sharing
[image]

g430s with same anti smoothing and area 
https://drive.google.com/open?id=1Z73CWIpzPoSAFDAjOOHELuuNhh8sfCT_
[image]

and just for added fun the pen swap 1060pen on g430s without antismoothing it really does compare to the the one with antismoothing when moving slow
https://drive.google.com/open?id=15gOSrDasgpSRu3eHViubHx6nOky9Ec4a
also had to disable buttons for added fun
[image]

closed 2018-12-16 by riley-badour