hawku/TabletDriver #307: the new 1060 plus button support

riley-badour, 2018-12-16:
## Hardware/software information
Windows Version: 

Tablet Driver version: 2.3

Tablet model: the new 1060 plus

got up to 6 buttons working

[debug1060buttons.txt](https://github.com/hawku/TabletDriver/files/2683896/debug1060buttons.txt)
it goes up to 6 buttons

1   2
4   8
10 20
other group of buttons are 
40 80 these are are the same source as above and they don't seem to work
1   2 theses are in source 5 instead of 4  
4   8
nothing that needs to be done in a hurry is just my tablet is a button heavy one 
and the button mapping is great

hawku, 2018-12-16:
Try these settings:
```
AuxReportId 0x08
AuxReportLength 12
AuxDetectMask 0xE0
AuxgnoreMask 0x80
AuxCustomData ReportId Source=0
AuxCustomData ButtonsLow Source=4
AuxCustomData ButtonsHigh Source=5
AuxCustomData Detect Source=1
AuxButtonCount 12
```

riley-badour, 2018-12-16:
prefect hawku  you can add that to the config for a later update

closed 2018-12-22 by hawku