hawku/TabletDriver #309: Devocubs Anti-chatter?

badaccuracy, 2018-12-18:
Would it be possible to add devocubs Anti-chatter in your driver? I can't seem to find settings that feel like my settings on the devocub Anti-chatter. I'm requesting this because your 0.2.x version has tablet button support and supports the XP-Pen g640s. If this doesn't belong here, you can close this issue. Thanks

## Hardware/software information
Windows Version: Win 10 x64

Tablet Driver version: 0.2.3

Tablet model: XP-pen g640s

hawku, 2018-12-18:
@badaccuracy Current implementation of Devocub's anti-chatter can't be added to the driver.
Are you really using anti-chatter on G640S?
That tablet already has enough hardware smoothing, so I don't recommend to add any extra latency.

If you have a Wacom tablet, try these settings:
- Smoothing filter 20 ms
- Noise filter 10 and 0.5 mm

badaccuracy, 2018-12-18:
Will try these settings. Thanks

closed 2018-12-22 by hawku