hawku/TabletDriver #314: Doesn't work with > 1 monitor plugged in

qewc, 2018-12-23:
Ok so basically, it cuts off halfway on the monitor i bind it to.

## Hardware/software information
Windows Version: 10 (LTSB)

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G430s

riley-badour, 2018-12-23:
can you give your startuplog.txt and a screenshot of your tablet driver area window to help ypu with your issue

hawku, 2018-12-25:
@qewc Windows might have problems with your monitor setup or you have manually set the desktop size to wrong values.

Set the desktop size to automatic. If that doesn't work, you can try SendInput (Debug) mode.

andyrkn, 2018-12-27:
this happens on nvidia optimus laptops

andyrkn, 2018-12-30:
@hawku have you tested setups that have two monitors with different  resolutions?

hawku, 2018-12-30:
@andyrkn Yes I have tested with multiple monitors with different resolutions, DPI values and refresh rates.
http://hwk.fi/TabletDriver/TabletDriverMultiResolution.png

The problem is with how Windows handles absolute positioning devices on some special cases which I can't reproduce.
SendInput (Debug) mode should work when the absolute positioning is broken.
If that doesn't work, then there isn't other way than to edit TabletDriver's desktop size values (Misc tab) to match the screen area that an absolute positioning device can reach.

closed 2018-12-30 by hawku